solosky/pixl.js #295: Option to emulate other figures like Disney Infinity, Lego Dimensions or Skylanders

maciak, 2024-11-30:
HI,
I see you have added the Card emulator and there are multiple cards to choose from. It would be good to do separate options for other figures like Disney Infinity, Lego Dimensions or Skylanders. Or if the custom card already does support them, a short instruction on how to use it as it's a bit confusing for me now.

skylandersNFC, 2025-04-17:
Ha, i see one of my resources above.

Actually i'm here since i've started testing the pixl js for Disney Infinity.
They are using Mifare Mini S20 7 bytes UID cards and I saw such inside the Card Emulation menu.
However, the Mifare Mini included seems to be 4 bytes UID and there is no way to change it to 7 bytes.

So far this is the one major block for Disney Infinity support.
If we get a 7 bytes UID support, it might actually work.

@solosky Do you think such feature is viable and can be easily implemented?
Gratitude

skylandersNFC, 2025-04-18:
It's me again.
Did some research regarding Lego Dimensions, and those are on NTAG213, which are also available for emulation from the Card Emulation menu.
**BUT**, they also require 7 byte UID.

So getting a firmware which supports 7 byte UID will unlock both Lego and Disney games all together.

solosky, 2025-04-20:
Hi @skylandersNFC , thanks for your feedback. 

### 7 bytes UID support for MF mini and NTAG213
This features are supported by low level function, but not enabled for UI setup, I will enable the 7 byte UID in next major release. 

Meanwhile, some idea on how to more friendly to emulate the different tag which should be help on emulation for Lego and Disney games.

### FlipperZero .nfc format support 
the .nfc file format is very simple and powerfull format for NFC data, support .nfc file format for Card Emulator could be easily load into Card Emulator and nfc data file generation. 

### Easy emulation by select bin/nfc file 
Card Emulator introduces a new mode that could select .nfc/.bin file to emulate directly, without load into slot of card emulator. 
the card data in card data buffer could be discard or save to origin file directly. 

Please share your thoughts on my idea, if you have more great idea, please kindly let me know.

skylandersNFC, 2025-04-20:
Thank you so much for this update. I'm super stoked to test it.

skylandersNFC, 2025-06-06:
@solosky Hi there, I don't mean to be a bother, but I was wondering if there's any estimated date for the upcoming major release ? Thank you!

skylandersNFC, 2025-06-10:
@solosky God damn, it works for Disney !
I will do a written documentation on it soon :)

solosky, 2025-06-10:
@skylandersNFC  good to know that! 
Without enable custom mode, you could choose the uid size to 7 bytes uid which will read uid from block 0. But the SAK is not stored block 0, it was hardcoded. 
The 10 bytes uid is not implemented because I cannot find such card to test.

solosky, 2025-06-11:
Hi @skylandersNFC  thanks for feedback. 
Regarding current commits, may I know is there any bug or improvement on this version? 
If you feel good to release current commits, I will release current commits as version 2.13.0. 
https://github.com/solosky/pixl.js/pull/328

skylandersNFC, 2025-06-11:
> Hi [@skylandersNFC](https://github.com/skylandersNFC) thanks for feedback. Regarding current commits, may I know is there any bug or improvement on this version? If you feel good to release current [...]

For Disney Infinity - everything is working with the new change. We have not tested any LEGO games yet.

However, for Disney the end user need to do the following steps:

1. Inside Card Emulator > Open Slot 1
2. Change card type to MiFare Mini
3. Go to Advanced menu > Change ID Size from 4 Bytes to 7 bytes
4. From Data.. > Load a Disney dump (usually a bin file)
5. Go back to Advanced > Change Custom Mode from [OFF] to [ON]
6. Find SAK value and change it to from "08" to "09"
7. Find ATQA value and change it from "00.04" to "00.44"
8. Go back to Custom Mode and disable it (from [ON] to [OFF])

Explanation: The Disney dump emulation would need SAK 09 and ATQA 4400, hence those have to be changed manually.
In order to do that, the custom mode have to be enabled. However, while [ON] the Custom Mod will overwrite the UID of the dump to some generic one. No idea why this feature is there. So in order to prevent this, Custom Mode have to be set to [OFF] to get the real dump UID while still keeping the changed SAK 09 and ATQA 4400

solosky, 2025-06-11:
HI @skylandersNFC , 

If you do not enable the Custom Mode, just select the correct UID size, the UID, ATQA is read from block 0. however, the SAK is not stored in block0, so it always use a hard coded value 0x08, which also can be changed in Custom Mode. 

But if enable the Custom Mode, UID, ATQA, SAK are read from dedicated storage other than block0 and the initial value was hardcoded, so you can custom everything in anti-col phrase. 

As a workaround, you could enable custom mode first and change the SAK first. then disable custom mode. the UID and ATQA is read from block0 and SAK is your customized value. 

The ATQA is read from 9,10 bytes in block0. it should be same as the actual card. I ready don't know why the ATQA in dump file is not same as game expected...

skylandersNFC, 2025-06-11:
> HI [@skylandersNFC](https://github.com/skylandersNFC) ,
>
> If you do not enable the Custom Mode, just select the correct UID size, the UID, ATQA is read from block 0. however, the SAK is not stored in [...]

Ahh, this makes sense.

The hex values on the dumps are "89 44 00 C2"
This corresponds to ATQA 4400 and SAK 08

From those only SAK have to be altered to 09, since this is what the game accepts.

But so far when using the Custom Mode to change the SAK, it was also altering my ATQA and UID to something else, so I had to go back and change it manually.

Now with the new firmware, after enabling the Custom mode and losing the proper ATQA and UID, I've used the "Load from Blok0" and indeed i got the correct UID and ATQA from the dump data.

Then i've changed SAK to 09 manually, while leaving the Custom Mode [ON] this time, and everything works as before.

Splendid

Is there a way to make it also read the SAK values from the dump data ?
I will alter the data on all Disney dumps just to have SAK 09 instead of 08, if this is possible.
Gratitude

solosky, 2025-06-12:
Hi @skylandersNFC , easy, I could make the SAK hardcoded to 08 if you click menu "Load from block0.."

solosky, 2025-06-12:
@skylandersNFC 
changes have be committed for bellow hardcoded sak when loading UID/ATQA/SAK from block 0 according to tag type:
MF mini: 09
MF 1K: 08
MF 2K: 19
MF 4K: 18

Reference: https://www.nxp.com/docs/en/application-note/AN10833.pdf
 
You could download latest firmware to test via Actions:  
https://github.com/solosky/pixl.js/actions/runs/15614134350

skylandersNFC, 2025-06-12:
[image]

Gratitude for the awesome firmware changes.

In several hours we will finish the Disney testing, just to wrap everything with it.

And then it's time for LEGO!

Since this ticket is for both (and even Skylanders), I will continue the discussion here.

TRD2B, 2025-07-23:
Good evening,
@solosky Any Plan to add menus for Games like Disney Infinity, Lego Dimensions and Skylanders etc... if so it would be huge, @skylandersNFC did you make the guide you were talking about, I would be very interested in reading/testing it and why not give my feedback on the accessibility and understanding of this guide.

skylandersNFC, 2025-07-25:
@TRD2B Now you can use official firmware for Skylanders and Disney Infinity without any problems.
When i was making the guide the Disney Infinity firmware changes were not yet public.

They are now after 2.13.0 - https://github.com/solosky/pixl.js/pull/328

And for Skylanders the custom firmware is just for more slots and if you are on Nintendo Switch, it saves you the hassle of changing the SAK values manually, which you can still do with official firmware as well.

Regarding LEGO, i'm still testing it.
The initial test failed, so i was wondering if the NTAG21x can emulation even works
https://github.com/solosky/pixl.js/issues/340
Don't know when I will be back on that tho, have a lot of other stuff lately.

TRD2B, 2025-07-25:
@skylandersNFC Thanks to you for these guides, this research work, it's fabulous and @solosky also thank you for making all these changes, I went up the thread of the discussion and can't wait to see if Lego Dimension manages to be emulated, it would be the ultimate device, thank you both and the community.

skylandersNFC, 2025-08-01:
@TRD2B I’m assuming you have a LEGO Dimensions portal and Allmiibo, since you showed interest here.
Could you reach out to me on Discord so we can go over some ways to test the new LEGO dumps on Allmiibo?
I’m on the [official Pixl.js channel](https://discord.gg/4mqeQwcAB2) under the same username. Thanks!

skylandersNFC, 2025-08-04:
@solosky Hi again,

We have several people meddling with LEGO and Allmibo now, and i've been told it works for them with the current firmware.
However, when the dump is loaded, it wont emulate the proper UID, which have to be added manually.

So it seems the NTAG family emulation is in a dare need of the new "Load from block0.." feature, or even better - if it's possible to just get the dump UID when the dump is loaded onto the slot ?

Thank you

skylandersNFC, 2025-08-05:
@solosky Just tested the latest dev build - works like a charm for NTAG's UIDs.

I will also ask around about the SAK and ATQA values.
The default ones are `00` and `44 00`.
Want to make sure those are proper for LEGO.
Will come back to you.
Thank you