solosky/pixl.js #302: How to create dump files for card emulation

gabrielmcf, 2024-12-26:
Sorry, created this with wrong user. Just copying from the other issue:
It was not clear to me how can I create a binary dump file of an NFC tag to use in the card emulator. I've tried a few android tools but none can give to me out of the box a binary format. Not sure how to convert the hex file to binary. Appreciate if we could add more details on that in the docs.

solosky, 2025-01-14:
What kind of card do you want to emulate in the card emulator app ?

Alex12kip, 2025-02-13:
As he dont seems to be responding anymore im just going to ask it, could you make a in depth tutorial for every kind of card? I have no idea what my card uses and how would i identify it? App on ios or something?