solosky/pixl.js #303: How to get amiibo data back after accidentally DELETING all data. Please help

Bruh123456780, 2025-01-01:


impeeza, 2025-01-01:
Just restoring a backup or copying all files manually, there is no «undelete» possible on that filesystem.

Bruh123456780, 2025-01-01:
How do I restore a backup? You gotta help me man, I’m not gonna waste 43 goddamn dollars

Bruh123456780, 2025-01-01:
How do I restore a backup or copy all files manually? Ble transfer isn't working for me. I need answers man

impeeza, 2025-01-03:
> How do I restore a backup or copy all files manually? Ble transfer isn't working for me. I need answers man

Happy New Year Mate,

you Allmiibo device legally, should come without any amiibo files, because that files are copyrighted material from big N.

On the reality most of us did received our devices with some set of files on it. it is illegal to any of us to share that files with anybody but you can use google for search «Amiibo Bin Drive V2!» and you can find a source to download that files.

Your device have some modes:
[image] «Amiibo emulator», on this mode you device read «.bin» files which are dumps of the original amiibo figurines, remember EACH figurine have a unique ID, this mode is intended for the device emulte YOUR ORIGINAL DUMPS of your own figurines, you can dump them using a phone with NFC capabilities or a Nintendo 3DS. this mode have the advantage of you are using your original ID so any data asociated to it is preserved on the amiibo and the consoles.

[image] «Amiibo database» on this mode the Allmiibo device uses a internal database to generate a «virtual» and «ephemeral» amiibo which always have a random ID and can not hold amiibo data.

[image] «Amiibo Link» on this mode the device emulate a amiibo link device and can be used like that device, for creating the amiibos and emulate the one selected, for this mode you need the correspondant app on your phone and bluetooth communication

[image] «Card Emulator» on this mode your device will emulate RFC cards using dump files or cards created from scratch using number data.

[image] «BLE File transfer» this mode is used to transfer files from/to the device using the Web or a phone app (different to the one for the Link Mode)

So please let us to know if this resolve your question.

Bruh123456780, 2025-01-03:
This has helped a lot thank you, one last thing, the allmiibo says "amiibo keys not loaded, upload the file key retail.bin to the root of-" can you tell me where I'm supposed to upload the file?

Bruh123456780, 2025-01-03:
Listen man you are a LIFESAVER. You basically saved me 43$. Thank you. Closing discussion

solosky, 2025-01-14:
> > How do I restore a backup or copy all files manually? Ble transfer isn't working for me. I need answers man
> 
> Happy New Year Mate,
> 
> you Allmiibo device legally, should come without any amiibo files, because that files are copyrighted material from big N.
> 
> On the reality most of us did received our devices with some set of files on it. it is illegal to any of us to share that files with anybody but you can use google for search «Amiibo Bin Drive V2!» and you can find a source to download that files.
> 
> Your device have some modes: ![image](https://private-user-images.githubusercontent.com/41979604/399843090-d89a53a5-2a7f-48a1-8ab7-a2211e2c1249.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY4MjI1NTcsIm5iZiI6MTczNjgyMjI1NywicGF0aCI6Ii80MTk3OTYwNC8zOTk4NDMwOTAtZDg5YTUzYTUtMmE3Zi00OGExLThhYjctYTIyMTFlMmMxMjQ5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTE0VDAyMzczN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJkOTkzYjEwNWFjMTc5YWU0ZmM5MzZkYzg5MDRkODJmMzAwNmQ1YjdlYjdkNzRiOTljMTNkMWEyZWE1MzI1ZDUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.sQjQFBLsvAbnEMHM6hhFLITwXx8Jroe-Gunx2cVEAJM) «Amiibo emulator», on this mode you device read «.bin» files which are dumps of the original amiibo figurines, remember EACH figurine have a unique ID, this mode is intended for the device emulate YOUR ORIGINAL DUMPS of your own figurines, you can dump them using a phone with NFC capabilities or a Nintendo 3DS. this mode have the advantage of you are using your original ID so any data associated to it is preserved on the amiibo and the consoles.
> 
> ![image](https://private-user-images.githubusercontent.com/41979604/399843369-7965dc35-7e07-4517-94d0-c98ebd633d49.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY4MjI1NTcsIm5iZiI6MTczNjgyMjI1NywicGF0aCI6Ii80MTk3OTYwNC8zOTk4NDMzNjktNzk2NWRjMzUtN2UwNy00NTE3LTk0ZDAtYzk4ZWJkNjMzZDQ5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTE0VDAyMzczN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVjMTFjN2NlM2RiZTY3MjRlMzc2NjQ4MjZlZmRjZTExNmNhZWIxMWUxODAxNDEyZjYyY2RlMDdmZDRkMWFhOGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.bP5n7KlwSXKYID2JuHi-2mm58xDmPNJS9gVIj9XBups) «Amiibo database» on this mode the Allmiibo device uses a internal database to generate a «virtual» and «ephemeral» amiibo which always have a random ID and can not hold amiibo data.
> 
> ![image](https://private-user-images.githubusercontent.com/41979604/399843694-908c708e-1272-4987-80eb-84ae7cb97a6c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY4MjI1NTcsIm5iZiI6MTczNjgyMjI1NywicGF0aCI6Ii80MTk3OTYwNC8zOTk4NDM2OTQtOTA4YzcwOGUtMTI3Mi00OTg3LTgwZWItODRhZTdjYjk3YTZjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTE0VDAyMzczN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThkNGJmNzBhZjk1MTRjY2QzMmQxODU5YjdlYzU2YmQ2MGI5NjBhZGM2MjI1NDRjYzM5MjY0NDY3ODkyMTY3ZjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.AoavvNalQndn4yHPRK8-dt8GCTqbAUMY04NRFQkUqyk) «Amiibo Link» on this mode the device emulate a amiibo link device and can be used like that device, for creating the amiibos and emulate the one selected, for this mode you need the correspondent app on your phone and Bluetooth communication
> 
> ![image](https://private-user-images.githubusercontent.com/41979604/399843850-65fbb064-ee30-4390-a43e-b1338a100716.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY4MjI1NTcsIm5iZiI6MTczNjgyMjI1NywicGF0aCI6Ii80MTk3OTYwNC8zOTk4NDM4NTAtNjVmYmIwNjQtZWUzMC00MzkwLWE0M2UtYjEzMzhhMTAwNzE2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTE0VDAyMzczN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTAzMjZhZTE4YTM2NWFlM2EzYWJkYjczMDhjN2NmNjA0MzgxNzhjNWU4N2FkNmRiMjkxMTc0ODZjM2JjMmMyMjYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.UQfPPy-brr9X42yVDWgaic2lCmKlZZx_Q7s1EZBNBWM) «Card Emulator» on this mode your device will emulate RFC cards using dump files or cards created from scratch using number data.
> 
> ![image](https://private-user-images.githubusercontent.com/41979604/399843942-c17c9ac9-8897-4e24-b787-639d647d4fed.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY4MjI1NTcsIm5iZiI6MTczNjgyMjI1NywicGF0aCI6Ii80MTk3OTYwNC8zOTk4NDM5NDItYzE3YzlhYzktODg5Ny00ZTI0LWI3ODctNjM5ZDY0N2Q0ZmVkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTE0VDAyMzczN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY3NmY4NmRhYTk3YjkyYWZiNDAyNGZjODAyNmIyMDIxNzc1NmY5NDJlZjFlOTE3MWQ4NGE4ZGRkZTYzN2QxYmImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.p4BclRWaqVMkcKrSkA0OHc5cxvp8dadQ2wyEu0uF8XE) «BLE File transfer» this mode is used to transfer files from/to the device using the Web or a phone app (different to the one for the Link Mode)
> 
> So please let us to know if this resolve your question.

it's a very detailed description on app features.  I will put it in the documentation later.

skylandersNFC, 2025-06-12:
Awesome explanation.

Just wanted to share that one of the allmiibo sellers actually provide a backup of the data file on their website:
https://allmiibo.com/support/

[image]