solosky/pixl.js #308: LCD 2.12.0 firmware do show nothing

shurillu, 2025-01-18:
Hello,

   I've downloaded the firmware for the LCD version from the release page (2.12.0) and programmed with the web flasher (https://thegecko.github.io/web-bluetooth-dfu/examples/web.html).
After flashing, the screen light turn on but without showing any menu.
Falling back with the trick ANY KEY - LEFT - MIDDLE - 5x LEFT - MIDDLE -> DFU MODE -> reflash 2.11.2 with the web flasher.

Any clue?

Thanks, 

Stefano

impeeza, 2025-01-18:
> Hello,
>
> I've downloaded the firmware for the LCD version from the release page (2.12.0) and programmed with the web flasher (https://thegecko.github.io/web-bluetooth-dfu/examples/web.html). After [...]

Yes read the Release notes (https://github.com/solosky/pixl.js/releases/tag/2.12.0):

For LCD versions, if the screen appears too faint after updating, you can reset the settings by following these button presses after the device enters standby mode: Middle, Left, Middle, Right (x3), Middle, Right (xN) until the screen becomes clear. Set the contrast to 80.

use the keys described to get the contrast high and the screen show text.

shurillu, 2025-01-19:
Thank you so much, it worked!

closed 2025-01-19 by shurillu

HorstHilch, 2025-01-21:
> Thank you so much, it worked!

Could you explain to me again how you did it? I followed impeeza's description but am I still stucked with the light turned on and no menu. However, I am sure I did something wrong on my side. 

When the screen turned off I press the following buttons: Middle, Left, Middle, Right (x3), Middle and then multiple times Right, but nothing happens.

shurillu, 2025-01-22:
Hello HorstHilch, 
  I did exactly the procedure explained by impeeza (display off -> MIDDLE -> LEFT -> MIDDLE -> 3x RIGHT -> MIDDLE -> Nx RIGHT)

You have to do the "Nx RIGHT" several times until the display start to show something (I don't remember but at least 5-10 times).
Anyway you can reflash the device with older firmware by enabling the DFU mode

MIDDLE - LEFT - MIDDLE - 5x LEFT - MIDDLE -> DFU MODE -> reflash 2.11.2 with the web flasher

Keep in mind that the DFU mode have a timeout; my advice is: prepare the web flasher with the correct file and after that enter in the DFU mode.

Good luck!

Stefano