solosky/pixl.js #309: Amiibopro screen freeze

Rogelio-dr, 2025-02-08:
So, my amiibopro froze out of nowhere, I actually just bought it.
I have no idea what happened, I was looking for a Mario ammibo to try.
I had already used it a couple of times, and it had worked correctly. I'm waiting for the battery to discharge but I would like to know what happened.
I got stuck on this screen:

[image]

solosky, 2025-02-10:
You may seek help from seller on how to reset the device. 
Some derivative versions of the devices come with built-in Hall sensors. Try placing your device near an iron-containing casing, such as a refrigerator, to see if it works.

grimson73, 2025-06-15:
I just tested 'my' oled derivative and it is rebooting when using a magnet:
Keep magnet close to back of device and then remove magnet -> device should reboot.
Added for anyone that stumbles on this topic.