solosky/pixl.js #313: Upgrade from 2.11.2 to 2.12.0 Fails on LCD Device

ecnepsnai, 2025-02-18:
Hi there! I have a LCD device and I cannot upgrade to the 2.12.0 LCD firmware. After upgrading using the nRF app on my iPhone, the device reboots and backlight turns on but there is nothing on the screen.

**However, I'm sure I have a LCD device** because if I use the 2.11.2 LCD firmware and upload that, it works fine - although it has very poor contrast, hence why I want to update to 2.12.0.

I've tried a few times now and it's always the same outcome, the upgrade seems to work but nothing is shown on the display.

[image]

MayanKoyote, 2025-02-18:
try to read release notes

ecnepsnai, 2025-02-18:
Oh, I'm sorry - I missed that bit.

> For LCD versions, if the screen appears too faint after updating, you can reset the settings by following these button presses after the device enters standby mode: Middle, Left, Middle, Right (x3), Middle, Right (xN) until the screen becomes clear. Set the contrast to 80.

This resolved my issue. Apologies for the noise!

closed 2025-02-18 by ecnepsnai