solosky/pixl.js #314: Empty screen after update 2.12.0

Askalive77, 2025-02-19:
Hello, sorry for my bad English. After updating to firmware 2.12.0 on my Allmiibo device, my screen is empty but the backlighting is on. I did use the LCD version of the firmware. Is my Allmiibo dead? Thanks for your help.

Askalive77, 2025-02-20:
Hello,

Sorry, I didn't see the following explanation: For LCD versions, if the screen appears too faint after updating, you can reset the settings by following these button presses after the device enters standby mode: Middle, Left, Middle, Right (x3), Middle, Right (xN) until the screen becomes clear. Set the contrast to 80.

I managed to adjust the contrast. 
The setting should be set to 50 or 60% by default.

SchemaMarkApp, 2025-04-12:
same here, just blank scree, not a contrast issue.

Anonmalies, 2025-05-17:
> Hello,
>
> Sorry, I didn't see the following explanation: For LCD versions, if the screen appears too faint after updating, you can reset the settings by following these button presses after the device [...]

Hi, thanks so much for the solution. I can confirm that this was indeed a contrast issue, for some reason, it was set to 0%. At first, I was afraid I might have bricked it, but fortunately, it turned out to be just a contrast setting problem.