solosky/pixl.js #318: Allmiibo Keeps Restarting

asterprism, 2025-03-09:
I just recently got the Allmiibo. I updated the firmware and loaded the missing amiibos I wanted onto the device. For a short time, everything was working fine, but now whenever I try to go through “Amiibo Emulator” and choose an amiibo to use, it restarts and goes back to the main menu as soon as you hit select on an amiibo. I really hope there’s a way to fix it

solosky, 2025-03-11:
Did this happen when click a specified amibo bin files or any amiibo bin files will case this issue ?

asterprism, 2025-03-12:
I only really use it for Animal Crossing and it seemingly does it for any in the main series

MysticalEternity, 2025-07-05:
I have the same issue when I try to select one of the amiibos under the "Mario Sport" folder