solosky/pixl.js #322: OLED屏幕在运行游戏的时候显示不对齐

x-magic, 2025-03-18:
游戏在OLED版上运行的时候会向左偏移2个像素，即游戏界面左侧少显示两列像素，右侧露出菜单的滚动条。很奇怪LCD版本不存在这个问题。

两台机器都重新线刷以排除固件损坏问题。如图：

[image]
[image]
[image]
[image]

solosky, 2025-04-20:
这个问题我也发现了，不过我可能不再更新游戏的内容了，因为目前的flash空间已经完全满了，下一个版本不会把游戏编译进默认固件，但是代码可以保留，可以自行编译固件。

x-magic, 2025-04-20:
> 这个问题我也发现了，不过我可能不再更新游戏的内容了，因为目前的flash空间已经完全满了，下一个版本不会把游戏编译进默认固件，但是代码可以保留，可以自行编译固件。

赞成，相比之下能放个GB2132字库进去还是更实用一些

closed 2025-04-20 by x-magic