solosky/pixl.js #323: DFU update glitch

ApertureLabs4194, 2025-04-09:
when updating the pixl.js over bluetooth, it gets stuck at "4096/372180 firmware bytes written" and the device says "dfu update waiting"

MayanKoyote, 2025-04-10:
In a decent society it is customary to specify the equipment used and attach logs. There are no fortune tellers with fortune balls here.

impeeza, 2025-04-11:
My cristal ball crashed!

solosky, 2025-04-20:
Try update the firmware with Mobile APP MToolsBLE, it has decend build-in support for firmware update for Pixl.js device.

Waitsnake, 2026-11-12:
I also got stuck with the **4096-byte hang issue** when trying to update my Pixl.js through the website using Chrome on my Mac.

However, I found the tool mentioned by @solosky: **MTools BLE RFID Reader** in the Apple App Store. Surprisingly, it even runs directly on my Mac (macOS 15), without needing an iPhone!

https://apps.apple.com/us/app/mtools-ble-rfid-reader/id1531345398

The app detected my Pixl.js over Bluetooth, even though it was still stuck in an endless DFU mode after the failed web update.

I was able to flash the firmware successfully using MTools BLE, and my device is now working normally again, running firmware **2.16.0**!

**Here's how to do it:**

1. Open MTools BLE.
2. Go to **Tools → Devices**.
3. Select your Pixl.js (mine was listed as `pixl.js Amiibo`).
4. Open **Firmware**.
5. Select the OTA ZIP file.
6. Choose your Pixl.js device.
7. Click **Start** and wait for the update to finish.

That's it! My Pixl.js was back to normal without opening the device or using an external programmer.

Hopefully this helps anyone else experiencing the dreaded `4096 bytes written` issue. Don't give up on your device.

Thanks to @solosky for recommending MTools BLE in the first place.