solosky/pixl.js #331: Ability for Allmiibo to recursively scan subfolders within the "chameleon/dumps" directory for use with the "Card Emulator".

skylandersNFC, 2025-06-13:
Greetings,

This ticked is tied to https://github.com/solosky/pixl.js/issues/330 with the file path length limitation.

Let's imagine the file path length limitation can be increased - in that case, would it be possible to add support for subfolders within the "chameleon/dumps" directory?

[image]

To clarify, I'm not requesting support for _uploading entire folders_.

Rather, the ability to organize dump files into multiple subfolders within "chameleon/dumps", and have the "Card Emulator" menu automatically detect and iterate over all files within those subdirectories.

Gratitude

solosky, 2025-06-13:
I think the possible solution is to make the dump file selection supports read and navigate in sub directory, like Amiibo Emulator do.

MayanKoyote, 2025-06-14:
Quick question about this new functionality:

`_L_OPEN_DIR_FAILED` ( "Open Directory Failed!" ) from here:
https://github.com/solosky/pixl.js/blob/a905509411c4304d6d8bff41efe9816a76186f84/fw/application/src/app/chameleon/scene/chameleon_scene_menu_card_data_file_load.c#L115

has the same meaning as the already existing `_L_OPEN_FOLDER_FAILED` ( "Failed to open folder" ) from here: ?
https://github.com/solosky/pixl.js/blob/cbe0566ad1ee40110074dba91940fec4df358752/fw/application/src/app/amiibo/scene/amiibo_scene_file_browser.c#L65

Maybe it's worth using the existing one ( `_L_OPEN_FOLDER_FAILED` ) in both cases?
Wouldn't that also help save some space?

solosky, 2025-06-15:
> Quick question about this new functionality:
> 
> `_L_OPEN_DIR_FAILED` ( "Open Directory Failed!" ) from here:
> 
> [pixl.js/fw/application/src/app/chameleon/scene/chameleon_scene_menu_card_data_file_load.c](https://github.com/solosky/pixl.js/blob/a905509411c4304d6d8bff41efe9816a76186f84/fw/application/src/app/chameleon/scene/chameleon_scene_menu_card_data_file_load.c#L115)
> 
> Line 115 in [a905509](/solosky/pixl.js/commit/a905509411c4304d6d8bff41efe9816a76186f84)
> 
>  mui_toast_view_show(app->p_toast_view, _T(OPEN_DIR_FAILED)); 
> has the same meaning as the already existing `_L_OPEN_FOLDER_FAILED` ( "Failed to open folder" ) from here: ?
> 
> [pixl.js/fw/application/src/app/amiibo/scene/amiibo_scene_file_browser.c](https://github.com/solosky/pixl.js/blob/cbe0566ad1ee40110074dba91940fec4df358752/fw/application/src/app/amiibo/scene/amiibo_scene_file_browser.c#L65)
> 
> Line 65 in [cbe0566](/solosky/pixl.js/commit/cbe0566ad1ee40110074dba91940fec4df358752)
> 
>  mui_list_view_add_item(app->p_list_view, ICON_ERROR, getLangString(_L_OPEN_FOLDER_FAILED), (void *)-1); 
> Maybe it's worth using the existing one ( `_L_OPEN_FOLDER_FAILED` ) in both cases? Wouldn't that also help save some space?

You are correct, I have made commit to use _L_OPEN_FOLDER_FAILED instead and removed _L_OPEN_FOLDER_FAILED

commit 78790366 (2025-06-15): Remove  _L_OPEN_DIR_FAILED, use _L_OPEN_FOLDER_FAILED instead. Fixed #331
closed 2025-06-25 by solosky