solosky/pixl.js #333: Safe to install v2.12.1 from iNFC?

grimson73, 2025-06-14:
Hi,

First thanks for all regarding developing firmware for this little device. Just received my OLED version and just tinkering with this :).

**Newer PIXL.JS firmware version listed on other site:**
https://github.com/strengthen/iNFCAmiibo/tree/main/Firmware/Pixl
iNFC (iOS) lists firmware v2.12.1 as an update available to the current installed v2.12.0.
As v2.12.1 isn't listed on this site as released i'm wandering if this version is safe to install?

In general, I did a compare with firmware 2.12.0 on both sites and they seem to be binary exact the same. Just out of curiosity does the other site mirror the firmware released from this site?

solosky, 2025-06-14:
Sorry, I did not release 2.12.1, I am not sure what the version came from. 
Please contact the repo owner for further information.

grimson73, 2025-06-14:
> Sorry, I did not release 2.12.1, I am not sure what the version came from. Please contact the repo owner for further information.

Thanks for answering, this might not be an appropriate question as this isn't related to this repository.
In essence I will only 'flash' firmware from this repository instead of the FW listed on the other repository to be safe.

closed 2025-06-14 by grimson73