solosky/pixl.js #343: Blank screen after 2.13.0

StarReaper3, 2025-06-28:
Hello. How do I revert back to a different firmware from 2.13.0? following the git hub guide does not seem to put the device into dfu mode.

DanCarrodeguas, 2025-06-28:
I ran into the same issue just a few hours ago after updating from version 2.11.2 to 2.13.0. The update appeared to complete successfully, but I ended up with a blank screen.

Here’s something you might try before reverting to an earlier version: what worked for me was following the steps found in the Release Notes for version 2.12.0, as referenced in issue [#341](https://github.com/solosky/pixl.js/issues/341). Applying those steps resolved the blank screen issue for me, although it took a couple of tries to get it right.

StarReaper3, 2025-06-28:
Update: got off the oled firmware by taking battery out and then pressing left twice then middle once to get into bluetooth.