solosky/pixl.js #351: Updated to 2.13 and everything is backwards

Techtronik, 2025-07-14:
just got one of these devices off Aliexpress and trying to get tags onto it but it wasn't working. So I tried doing a system update to 2.13 and now all the text and icons in every menu are backwards.

[image]
[image]
[image]

Techtronik, 2025-07-14:
tried downgrading to 2.12 and now it's stuck in the "DFU Update Waiting" screen. I can't do anything to get out of it and it's not letting me start another update.

Techtronik, 2025-07-14:
got it to install 2.11.2 and now it's working again

Techtronik, 2025-07-14:
Connected once and now it won't go past the data folder

[image]

Techtronik, 2025-07-14:
and none of the amiibo I managed to upload work. I tried them in BOTW and they all just made an error noise while the few of the amiibo I actually have worked fine regardless of whether or not they are from LoZ

cccccraneo, 2025-07-15:
You sure you used the correct update version for your device?
I applied the LCD update to my OLED device by error and the text started to be shown backwards until I applied the correct OLED update.
Try to use the other file, hope this helps.

Techtronik, 2025-07-15:
that could be what caused that. wasn't sure which one this device was.