solosky/pixl.js #353: Create a specific menu that includes the right settings for each game that can be emulated Lego Dimension/Skylander/Disney Infinity

TRD2B, 2025-07-24:
Bonjour @solosky ,
Je me demandais s'il serait possible de créer un menu pour chaque jeux pour lesquels les figurines peuvent être emulées à côté du menu Amiibo Emulator, un menu Skylander Emulator, un autre Disney Infinity Emulator et un dernier Lego Dimension Emulator qui incluerais directement les bons paramètres pour chaque type de figurines, si cela n'est pas trop complexe niveau code, utiliser vous des paramètres globaux ou des paramètres dédiés à chaque fonction, 
Merci d'avance.

TRD2B, 2025-07-29:
Good evening,
Thanks for answering, but skylandersNFC explained and directed me to its tutorials for Disney Infinity and Skylanders. Only Lego Dimensions is missing, and it will be the ultimate device. Good idea to add a quick emulation function.
Looking forward to testing the next update.
Thanks again.

skylandersNFC, 2025-08-05:
LEGO Dimensions is nearly finished, since we have some working attempts already and soon I will make a full tutorial on it.
@TRD2B I think you also said you found a LEGO toypad, so i will reach out to you for even more testing :P
It seems this ticket can be closed.
Cheers