solosky/pixl.js #354: Tried to update to 2.13.0 and file dosent work

Okamiryuku, 2025-07-26:
I have try to update to version 2.13.0, via Android app and web page updater, the web page give me this error when uploding the zip file : No file chosen Error: Unable to find manifest, is this a proper DFU package?, and the android app say init not found. I'm downloding the [2.13.0_LCD.zip] someone else having this problem or am i doing somethin wrong?

MayanKoyote, 2025-07-26:
reread this
mindfully
https://github.com/solosky/pixl.js/blob/main/docs/en/02-Flash-Firmware.md

Aeitto, 2025-08-01:
> I have try to update to version 2.13.0, via Android app and web page updater, the web page give me this error when uploding the zip file : No file chosen Error: Unable to find manifest, is this a [...]

Try this app: https://pixeldrain.com/d/ytTtMMxe
I got it from baidu and its a pain to download so u uploaded it on my cloud.
It works flawlessly, its a 3 step process. Remember to extract either LCD or OLED zip and pick the pixjs_ota file